usbcnc/grbl #1: Hope to post issues in here instead of main branch.

usbcnc, 2017-01-04:
If you have any issues regarding this port, please post here instead go to main branch's ARM discussing.

ewidance, 2017-02-20:
I don't know if this binay have been built with -DLEDBLINK -DUSEUSB, but i can't enumerate device while connected to USB on OSX with lsusb. Additionnaly, led is always ON....

usbcnc, 2017-02-20:
Can you try on Windows first? I have not tried this on OSX,

usbcnc, 2017-02-20:
Which driver do you use? Microsoft or ST's VCP?

usbcnc, 2017-02-20:
@ewidance Do you mind me take a look of your setup via teamviewer?  I am interested to see others using this build.

ewidance, 2017-02-22:
Ok i found it. I was using a 'bluepill', with the clock jumper removed. Board was rnning with internal rc osc @wrong frequency. After putting back the jumper port is properly enumarated under windows 10 (VM) & OSX (Sierra). Thanks a lot.

usbcnc, 2017-02-22:
Great!.

closed 2017-02-22 by usbcnc

usbcnc, 2017-02-22:
I can only see two jumpers and both are for the boot select.

ewidance, 2017-02-22:
It not exactly a blue pill.  It's a STM32F103C8T6 STM32 Minimum System Development Board
[image]