usbcnc/grbl #2: chip name problem

yaroslavVl, 2017-02-16:
did not find stm32f103c8 chip in coide 2.0.6 (Build id: 20170117-2.0.6). Which version do you use? I suppose it have to be stm32f103c8t6 to work well.
[image]
[image]
my other projects in stm32f103c8t6 (the board is the same as yours) are opened and work.

yaroslavVl, 2017-02-16:
Ok, I changed it only in stm32grbl11d.coproj, and project is opened

closed 2017-02-16 by yaroslavVl

SailWithChips, 2018-03-02:
Hello, same problem like you. Where do you changed it?. Plase could you attach some photo or something to know what to do. Thanks in advanced,