usbcnc/grbl #3: Possible incomatibily with Coide 2.0.6

ewidance, 2017-02-19:
I tried to build project with CoIDE 2.06. stm32grbl11d.coproj needs to be adapted since Coide needs now complete device name : "STM32F103C8T6" instead of "STM32F103C8" for 1.x version . Changes must be done at 2 places in the file, and needs to be edited BEFORE opening it with CoIDE.
Defines section needs also to be adated by ADDING (not replacing) a -DSTM32F103C8T6 .

Now the project builds.. but USB is not enumerated as serial port.

usbcnc, 2017-02-20:
I will try to use newer ide myself and let you know the result.

ewidance, 2017-02-22:
After properly configured clock jumper on my bluepill, serial port is enumarated with the builf done with joined files. 
Grbl banner is displaying properly on USB VCP. 

Now, I need to test with a cnc to completely verify  coocox 2.X build, but it seems to be promising.

usbcnc, 2017-03-04:
I downloaded CoIDE 2.06 and I modified the project file. The problem is when I used CoIDE 1.7.8 my binary size is about 37-38KB. However CoIDE 2.06 generate binary of 44.9K. Both with max optimization of size and no debug.

usbcnc, 2017-03-04:
After I added -ffunction-sections for compile, I got the same file size.

closed 2017-03-04 by usbcnc

SailWithChips, 2018-03-02:
Hello I've been trying to have something running on my STM32F103C8T6, compiling it with CoIDE 2.0.7. GCC tool is the latest one 7 2017-q4. Once I download "hex" file to the board, then the LED is allways "ON" and no serial communication with the board can be done. Could someone help me?. Thanks a lot in advanced.

usbcnc, 2018-03-02:
@SailWithChips . This could be the hardware. The resistor value is wrong. To work around it 
1. Try another PC.
2. Use a USB hub. 
3. Replace the resistor (google to find out how).

SailWithChips, 2018-03-02:
Thanks for your answer. 1. Same PC downloading different program working fine, for instance version 0.9 of grbl.
2. I expect that 1 is enough for determining not a problem on PC/Hub.
3. Already changed resistor to have 1.5 kOhm... 

Could someone place compiled working ‘hex’ somewhere?
Thanks again for answering!