usbcnc/grbl #4: No Makefile for command line building (Osx, Linux)

ewidance, 2017-02-19:
The Makefile included in the project is the one for building AVR Project. I've done a first port of that file. Additionnaliy, a linker script have to be included. 

Please find included 2 txt files (remove txt extention) as a suggestion for Makefile and Linker script.

[Makefile.txt](https://github.com/usbcnc/grbl/files/786202/Makefile.txt)
[STM32F103C8T6.ld.txt](https://github.com/usbcnc/grbl/files/786186/STM32F103C8T6.ld.txt)

usbcnc, 2017-05-12:
@ewidance ,By slightly modify your make file, I can build it under Linux.

misan, 2017-05-12:
@usbcnc Can you please share the modified Makefile?

usbcnc, 2017-05-13:
@misan Try it. 
I think this can also be used for cygwin under Windows.
[Makefile.zip](https://github.com/usbcnc/grbl/files/998250/Makefile.zip)

misan, 2017-05-13:
@usbcnc Thanks, I can confirm compile and link works with Linux (using Ubuntu 14.04 and ARM gcc 4.8)

usbcnc, 2017-05-14:
I change the first line to XTOOLS_DIR = /cygdrive/d/Program\ Files\ \(x86\)/GNU\ Tools\ ARM\ Embedded/4.9\ 2015q3 and I can use it under cygwin.

Also add CFLAGS += -ffunction-sections reduce the binary size significantly.

misan, 2017-05-14:
Wow, -ffunctions-sectios shaved off 8K from binary size 46K to 38K

MoonCactus, 2017-05-17:
I confirm it builds like a charm, thanks (did not try it in real yet though).

closed 2017-06-01 by usbcnc