usbcnc/grbl #11: Additional A B C axis

yaroslavVl, 2017-05-03:
Hello!
I've added additional axes.
The 4 axles work! 5 and 6 havn't checked yet.
So far, it has not done thorough testing (including limits, max step frec etc.).
I would like to clarify where the delay time (the frequency of updating the output STEP signals) is set (tune), because 4 axes output frequency will be lower  than for 3 axis, and for 6 - more lower, right?
So, where can I change maximum step rate?, for example, set the maximum output frequency for 4 axes  two times lower than now for 3 axis, to be sure that all operations will succeed.
One more question: whether it is necessary to expand any buffers, if so, which ones.
In the attachment, a project with modifications for 4 axes (without "grblwin" folder because of it size).
In the config.h (line 51) - // Define additional A, B and C axis

Sincerely, Yaroslav.
[grbl-edge ABC.zip](https://github.com/usbcnc/grbl/files/974840/grbl-edge.ABC.zip)
[image]

yaroslavVl, 2017-05-04:
I suppose these are TIM2 and TIM3, and TIM_Configuration function, but there are no comments, so I can't figure this out quickly. Can you explain these settings a little?

usbcnc, 2017-05-04:
I will take look latter. The max clock was I increase until it crashes. So nothing to be set. With more functions the max clock rate might be smaller and that is to be expected. If we need higher clock rate we can use STM32F4 CPU.
I agree we can increase the buffer since we have a lot more memory. However it depends on if the change will improve the performance.

yaroslavVl, 2017-05-04:
Many thanks!
You mean that there is no software limit for the maximum step signals rate (?)
Just if the output signals frequency is high and all tasks can not be handled, everything will crash?

I have not figured out the code yet.
I assumed that the timer sets a certain frequency of updating the output signals.

usbcnc, 2017-05-12:
The code looks good.

closed 2017-05-12 by usbcnc