usbcnc/grbl #13: Homing issue: Alarm 7

Elvenblood, 2017-05-28:
Has anyone tried homing and knows if it works?

I have limit triggers that I know works with the atmega-version. So the hardware should not be a problem. 

When I view "?" output it reports the triggers correctly in all axis.
I have the triggers at a high level and goes to a low when triggered.
As soon as I try the $H it moves the axis somewhat, locks up and then reports Alarm 7.

Elvenblood, 2017-06-01:
Well, I've tried lots of more things now, and I still don't know what the problem is. I've tried @yaroslavVl
port as well and it shows the same problem. The homing section works but after the STATUS_OK in the function system_execute_line the error ALARM: 7 occurs. I realize that it's probably something that needs to be reset. @usbcnc  do you think you can help me?

Thanks.
/E

usbcnc, 2017-06-01:
I think this can be debugged with the help of win32. Can you leave the contact info?

Elvenblood, 2017-06-01:
Sure my mail is elvenblood(at)hotmail.com. Just tell me what info you want and i will do my best to provide it. I dont have VS21015 installed right now. But will install it as soon as i can.

Thanks again!
/E

yaroslavVl, 2017-06-01:
offtop: it's not my port, it's @usbcnc port )))

yaroslavVl, 2017-06-01:
I'll try to look too.

Elvenblood, 2017-06-01:
Yes, I know its his port :) I just meant the code that you provided with more axis.

Thanks for looking into this as well @yaroslavVI.

yaroslavVl, 2017-06-02:
Usbcnc/grbl from box (only with limits remap 12,13,14 instead 10,11,12): when $22=1, homing is ok, when both $21,22=1 - alarm7.

Another problems: m8 (pb2) - ok, but remaped to pb4 - doesn't work. M7 doesn't work (defined in conf). Inputs: door(defined in conf), reset, hold, start - controller doesn't respond anymore until reset btn or repowered. pa7 pin work same as pa6 (DISABLE), I suppose it's elecrtical nois. There are more problems with remap.

usbcnc, 2017-06-02:
I will hook up the limit switch myself. I think there is a need to remap since I plan to add LCD/Keys/SD Card to the project. But I will not be able to to work on it until end of the month since I am traveling.

Elvenblood, 2017-06-03:
Well I've kind of figured out the problem. It all comes down to the interrupt routine EXTI15_10_IRQHandler.

I've used @yaroslavVl source code since I also want to use a 4th axis. But the problem should be the same I think.

I tried to change it to use enable and disable (NVIC_DisableIRQ/NVIC_EnableIRQ) That didn't work, and as I understand it disabling IRQ for a longer period is not a good idea. I kept my enable funtion but added a state varible to keep track if the interrupt should fire or not. This works and does not fire any unwanted interrupts. So my home routine now works.

I will also be out travelling for a while now, but as soon as I'm back I will try to look at this again.

/E

yaroslavVl, 2017-06-05:
could you share the source?

Elvenblood, 2017-06-05:
I have no github repository, but the changes are trivial. See below for more info.

I did not find a good way to empty the interrupts-queue when it was disabled, so as soon as it was enabled again the interrupts from the homeing got triggered. At least thats what I think.

I had to go back to @usbcnc's code though, since there seems to be some problem with the eeprom part of the 4-axis version. I haven't had time to look at it, but the problem is that when I try to zero a axis the coords don't get written to the eeprom. So the WPos and MPos stays the same.
I'll take a look at it when I'm back home again.

In limits.h I've added:
```
volatile uint8_t enable_when_done;
void limits_enable();
```

In limits.c:
Added the following to function limits_init():
```
enable_when_done = 1;
```

In limits_disable():
```
#ifdef STM32F103C8
  enable_when_done = 0;
  NVIC_ClearPendingIRQ(EXTI15_10_IRQn); //Probably not needed

  //NVIC_DisableIRQ(EXTI15_10_IRQn);
#endif
```

below disable:
```
#ifdef STM32F103C8
void limits_enable(){
  NVIC_ClearPendingIRQ(EXTI15_10_IRQn); //Probably not needed
  enable_when_done = 1;
}
#endif
```

in EXTI15_10_IRQHandler:
```
if(enable_when_done == 0){
  EXTI_ClearITPendingBit(EXTI_Line12 | EXTI_Line13 | EXTI_Line14 | EXTI_Line15);
  NVIC_ClearPendingIRQ(EXTI15_10_IRQn);
  return;
}
```

In motion_control.c, function mc_homing_cycle: (at the end)
```
#ifdef STM32F103C8
  limits_enable();
#else
  limits_init();
#endif
```

yaroslavVl, 2017-06-05:
>  when I try to zero a axis the coords don't get written to the eeprom
What do you mean "zero"? I've tried "G10 L2 P1 X10 Y10 Z10 A10", then "$#" and "?" all is ok.

Elvenblood, 2017-06-05:
Yes, that is what I meant. Strange that it doesn't work for me. I will try it more when I can. Have you tried the changes I proposed? If it's unclear what I mean just ask and I can try to explain it better.

yaroslavVl, 2017-06-05:
I try to do it now.
note: if you want to you need to use G53, you have to include the G53 on every line you want to use machine coordinates.
one more thing (I'll open another issue): To enable PB3, PB4 and PA15 pins you have to disable JTAG (SOLVED)

yaroslavVl, 2017-06-05:
When you build your project it's good to rebuild it completely (I had a trouble with "eeprom" (flash) when just Build. Sometimes I even use st-link utility to full erase flash (perhaps just prejudice)))))).

Elvenblood, 2017-06-05:
Nope, I've had the exact same problem. Now I always do rebuild to be sure :)

Elvenblood, 2017-06-05:
Do you use the CoIDE? and if so what version and which toolchain?
I've used the 1.7.8 with gcc-arm-none-eabi-6-2017-q1. But yesterday I tried EmBitz 1.11 with success.
Embitz was really easy to use with the debugger and st-link to set breakpoints when running my machine.

yaroslavVl, 2017-06-05:
CoIDE v2 beta, same gcc version

yaroslavVl, 2017-06-05:
What if just: `EXTI_ClearITPendingBit(EXTI_Line12 | EXTI_Line13 | EXTI_Line14 | EXTI_Line15);`?

usbcnc, 2017-06-26:
I have tried and it works fine for me for homing. Here is my seetings.
ok
$0 = 10    (Step pulse time, microseconds)
$1 = 0    (Step idle delay, milliseconds)
$2 = 7    (Step pulse invert, mask)
$3 = 7    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 2.000    (Homing switch pull-off distance, millimeters)
$30 = 100    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 1    (Laser-mode enable, boolean)
$100 = 400.000    (X-axis travel resolution, step/mm)
$101 = 400.000    (Y-axis travel resolution, step/mm)
$102 = 400.000    (Z-axis travel resolution, step/mm)
$110 = 600.000    (X-axis maximum rate, mm/min)
$111 = 600.000    (Y-axis maximum rate, mm/min)
$112 = 600.000    (Z-axis maximum rate, mm/min)
$120 = 45.000    (X-axis acceleration, mm/sec^2)
$121 = 45.000    (Y-axis acceleration, mm/sec^2)
$122 = 45.000    (Z-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 300.000    (Y-axis maximum travel, millimeters)
$132 = 50.000    (Z-axis maximum travel, millimeters)
ok
Error while processing response <ok>: An unexpected command was completed by the controller.>>> $X
[MSG:Caution: Unlocked]
ok
>>> $H
ok
>>> G10 P0 L20 X0 Y0 Z0
ok

usbcnc, 2017-06-26:
#define LIMIT_PIN        GPIOB
#define LIMIT_PORT       GPIOB
#define RCC_LIMIT_PORT   RCC_APB2Periph_GPIOB
#define GPIO_LIMIT_PORT  GPIO_PortSourceGPIOB
#define X_LIMIT_BIT      10  
#define Y_LIMIT_BIT      11  
#define Z_LIMIT_BIT      12  
 from CPU_Map.

usbcnc, 2017-06-26:
OK, I see if $21=1 I got Alarm7.

usbcnc, 2017-06-26:
Modify mc_homing_cycle at the end 
#ifdef STM32F103C8
	EXTI_ClearITPendingBit((1 << X_LIMIT_BIT) | (1 << Y_LIMIT_BIT) | (1 << Z_LIMIT_BIT));
	NVIC_ClearPendingIRQ(EXTI15_10_IRQn);
	NVIC_EnableIRQ(EXTI15_10_IRQn);
#else
  limits_init();
#endif
Fixed the issue.
@yaroslavVl  Please try my fix to see if it works.

yaroslavVl, 2017-06-26:
I'll try it

closed 2017-06-27 by usbcnc