usbcnc/grbl #14: Arduino IDE environment

mkeyno, 2017-06-04:
thanks @usbcnc for porting to such platform , however I've tried to change it and uploaded to my board regards to instructions , I think it little be handy , do you have plan to port it base on [Roger Clarck repo](https://github.com/rogerclarkmelbourne/Arduino_STM32),  its very stable now ?

usbcnc, 2017-06-04:
No, I am not Arduino guy myself.

mkeyno, 2017-06-09:
@usbcnc  why cpu_map.h is empty? have anybody use your lib for CNC laser?

yaroslavVl, 2017-06-11:
What do you mean "empty"?

yaroslavVl, 2017-06-11:
What do you mean "empty"?
https://github.com/usbcnc/grbl/blob/edge/grbl/cpu_map.h

usbcnc, 2017-06-12:
I do not consider this file is empty. 
I a out of town. I have ordered a laser module a while ago but have not arrived yet. I will test laser mode after I got it.

mkeyno, 2017-06-14:
I couldn't find couple of file like cpu_map.h  in coocox project tab or any other tab!
@usbcnc  when I install ST-link adapter , it just make installation in hardware usb category and not port category (STMicroelectronics STLink dongle) and not let the flash begin
```
Error: Connect failed, check config and cable connection
```

usbcnc, 2017-06-14:
I see what you mean about cpu_map.h. In the project if the .h file is specified you can easily open to edit. However it will not affect the compilation. You can still open the file from file menu.
STlink is not in port category. (like COM). You should better copy and paste your screen of device manager to see what is going on.

mkeyno, 2017-06-14:
please check the image 
[image]

usbcnc, 2017-06-27:
I have checked in project with cpu_map.h

closed 2017-06-27 by usbcnc

mkeyno, 2017-06-27:
@usbcnc  can I have you advise why STLink dongle not appear in port category ? also as previous discussion I was hopping I have your STM pinout for CNC laser 
thanks

usbcnc, 2017-06-27:
STLink is not a comport. You see this under Universal Serial Bus devices.
I will update the laser stuff once I have a changce to play it myself.

mkeyno, 2017-06-29:
I know that but the coocox seek the com port to upload the hex or I definitely miss somethings ?

usbcnc, 2017-06-29:
You need to setup properly on coocox. Setup teamviewer and I can help you out.
You can also use ST tool to program the target.

misan, 2017-06-29:
CoFlash tool worked for me too. http://www.coocox.org/software/CoFlash.php