usbcnc/grbl #15: PB3, PB4 and PA15 pins doesn't work (SOLVED)

yaroslavVl, 2017-06-05:
To enable PB3, PB4 and PA15 pins you have to disable JTAG:
GPIO_PinRemapConfig(GPIO_Remap_SWJ_JTAGDisable, ENABLE);

usbcnc, 2017-06-27:
I checked in the code in main.c

closed 2017-06-27 by usbcnc

alk1913, 2017-07-02:
hello,i am very excited whit this for of grbl.i am a user of grbl  in arduino,bat my experience whit this board(STM32f103 is really good ,i get the spindle PWM wor nice,and the coolant flo0d (PB3),isn t work,i remap tp PA9 and on this moment i not have troubles!!!THANKS ALL

SailWithChips, 2018-03-06:
Hello, thanks a lot for your work.
I've done same like **alk1913** and working succesful but I'm not able to have PA15 as probe input working as expected:
I added on main.c (line 92 - 95):
`#if defined (STM32F103C8)
    GPIO_PinRemapConfig(GPIO_Remap_SWJ_Disable, ENABLE);`

Also tested with (and not succesful):
 `GPIO_PinRemapConfig(GPIO_Remap_SWJ_JTAGDisable, ENABLE);`

I have allways fixed 3.17 volts on PA15... could anyone help me with this?.

Thanks a lot in advanced,