usbcnc/grbl #16: door, reset, feed hold and start inputs doesn't work

yaroslavVl, 2017-06-05:
system.c
WORK (modified code):
```
void EXTI9_5_IRQHandler(void)
{
	EXTI_ClearITPendingBit(EXTI_Line5 | EXTI_Line6 | EXTI_Line7 | EXTI_Line8);
	uint8_t pin = system_control_get_state();
	if (pin)
	{
		if (bit_istrue(pin,CONTROL_PIN_INDEX_RESET))
		{
			mc_reset();
		}
		else if (bit_istrue(pin, CONTROL_PIN_INDEX_CYCLE_START))
		{
			bit_true(sys_rt_exec_state, EXEC_CYCLE_START);
		}
		else if (bit_istrue(pin, CONTROL_PIN_INDEX_FEED_HOLD))
		{
			bit_true(sys_rt_exec_state, EXEC_FEED_HOLD);
		}
#ifdef ENABLE_SAFETY_DOOR_INPUT_PIN
		else if (bit_istrue(pin, CONTROL_PIN_INDEX_SAFETY_DOOR))
		{
			bit_true(sys_rt_exec_state, EXEC_SAFETY_DOOR);
		}
#endif
		NVIC_ClearPendingIRQ(EXTI9_5_IRQn);
	}
}
```
doesn't work (The original version):
```
void EXTI9_5_IRQHandler(void)
{
	uint8_t pin = system_control_get_state();
	if (pin) 
	{ 
		if (bit_istrue(pin,CONTROL_PIN_INDEX_RESET)) 
		{
			mc_reset();
		}
		else if (bit_istrue(pin, CONTROL_PIN_INDEX_CYCLE_START))
		{
			bit_true(sys_rt_exec_state, EXEC_CYCLE_START);
		}
#ifndef ENABLE_SAFETY_DOOR_INPUT_PIN
		else if (bit_istrue(pin, CONTROL_PIN_INDEX_FEED_HOLD))
		{
			bit_true(sys_rt_exec_state, EXEC_FEED_HOLD);
		}
#else
		else if (bit_istrue(pin, CONTROL_PIN_INDEX_SAFETY_DOOR))
		{
			bit_true(sys_rt_exec_state, EXEC_SAFETY_DOOR);
		}
#endif
		NVIC_ClearPendingIRQ(EXTI9_5_IRQn);
}
}
```

yaroslavVl, 2017-06-05:
solved?

usbcnc, 2017-06-06:
I am out of town and will take a closer look after I come back.

yaroslavVl, 2017-06-06:
ok

usbcnc, 2017-06-27:
I checked in change.
EXTI_ClearITPendingBit((1 << CONTROL_RESET_BIT) | (1 << CONTROL_FEED_HOLD_BIT) | (1 << CONTROL_CYCLE_START_BIT) | (1 << CONTROL_SAFETY_DOOR_BIT));

closed 2017-06-27 by usbcnc