usbcnc/grbl #18: pinout for cnc laser

mkeyno, 2017-06-14:
sorry to open ticket for my question but actually I've tired to build cnc laser controller with Arduino compatible  STF103CT6 board , however I've bit lost to assign the pins to the boards ,little hint really appreciated

mkeyno, 2017-06-17:
@usbcnc  I really appreciated for you hint

usbcnc, 2017-06-20:
I will soon get a laser module myself. I will see how to put it together.

mkeyno, 2017-06-21:
@usbcnc can you  direct me to the file with pinout  definition

usbcnc, 2017-06-29:
The control pin for laser is pin A8.  So far I am using $32=0.

usbcnc, 2017-07-03:
Check video. https://www.youtube.com/watch?v=CQwOXgtFz2o&t=6s

closed 2017-07-03 by usbcnc

mkeyno, 2017-07-09:
thanks for video , just for confirmation please check following image if it's correct
[image]

usbcnc, 2017-07-09:
@mkeyno PA9 and PA10 are not used for serial port. I use USB virtual COM port. So USB is not only power but also the main communication port to PC. Initially I found a need to use serial port for debugging. The SWD debugging was not working when USB is enabled. I found SWD debugging not reliable and use Windows is what I do most of the time to debug. 
Your spindle direction and spindle enable is different than mone, Mine was set to PB13 and PB14. Again you can find all the definition in cpu_map.h other than the PA9 and PA10 for serial port which is no longer needed.

mkeyno, 2017-07-11:
thanks @usbcnc  actually I found this configuration in cpu_map.h but not sure I referring to correct source, however I use bluetooth or wifi as serial bridge for every cnc I made so USB link is not helpful , can you give me advise what pins are better to use for wireless connection
BR