usbcnc/grbl #33: What is the step rate of grbl STM32?

radiolee, 2018-01-01:
We know grbl@arduino, pulse feq is about 30khz,  what now upgrade to 32bit@72MHz?

usbcnc, 2018-01-02:
I tried to 180KHz without issue.

closed 2018-01-02 by usbcnc

radiolee, 2018-01-07:
Thanks, this is great.

usbcnc, 2018-01-07:
If you need higher rate, you can.
1 Over clock the CPU. Then you might not be able to use USB but you can use serial port.
2. Use STM32F4 which runs at 168MHZ. So I assume the step rate can be easily go to 500KHz

radiolee, 2018-01-07:
Thanks for reply.  About 200kHz is suitable, lots of stepper driver working around this rate. BTW, is serial port working with USB enable? or must disable USB then serial is working? 
I just want connect a ESP8266 websocket module, is usb and serial working together or how to switch it quickly?

radiolee, 2018-01-08:
@usbcnc about 200kHz step rate is great for many stepper driver.