usbcnc/grbl #35: How to toggle USB VCP and serialport by an IO pin?

radiolee, 2018-01-23:
USEUSB was defined symbols in configuration, there is still free IO pins left.
It's because I want try ESP8266 websocket connection and keep USB VCP support, toggle it by a jumpper or a switch on the panel. @usbcnc could you support it as enhancement? or the both port working together maybe a good idea?

usbcnc, 2018-01-23:
What is the use of ESP8266 on Grbl? I do not have tested this module.

misan, 2018-01-23:
ESP8266 can be used as a serial-to-wifi module using ESP-LINK software for example.
https://github.com/jeelabs/esp-link

radiolee, 2018-01-24:
Yes, the ESP module is less than $2, run at 160MHz, it's create web socket connection via wifi for grbl. 
https://github.com/LaserWeb/LaserWeb3-ESP8266

radiolee, 2018-02-01:
I am close this because there is a easy way to use serial, just re-flash stm.