usbcnc/grbl #36: Spindle enable pin issue

lakeroe, 2018-02-02:
Hello,

for my application I need a separate spindle enable pin. With default settings PB13 is not working as expected. For this reason I uncommented following line in config.h
#define USE_SPINDLE_DIR_AS_ENABLE_PIN
This leads to an compilation error and I had to add "&& !defined(CPU_MAP_STM32F103)" in line 123 in grbl.h
Now it compiles fine and PB13 works as expected.
I just don't know if this is the intended way to do ...

Best regards,
lakeroe

SailWithChips, 2018-03-05:
I have same problem like you. But also with your lines I have not PB13 working as expected.

lakeroe, 2018-03-07:
My working BIN-File is attached (rename *.txt to *.bin). Try it if you like ...
[grbl.txt](https://github.com/usbcnc/grbl/files/1788254/grbl.txt)

SailWithChips, 2018-03-07:
Thanks for sharing, works your bin file. PB13 as expected, PA15 as probe input also fine now I would like to know in your BIN file were it's the PWM signal for spindle rpm's?.
Would be really appreciatted if you share also the CooIde project or at least the cpu_map.h ;)

Thanks a lot!!

lakeroe, 2018-03-07:
PWM signal is PA8.
The complete project is attached. I don't use any IDE, compile project by running _MAKE.bat

[GRBL.zip](https://github.com/usbcnc/grbl/files/1790226/GRBL.zip)

SailWithChips, 2018-03-08:
Thanks a lot lakeroe. With your bin and CPU_MAP, everything is working so nice... thanks a lot for your help!

closed 2018-03-23 by usbcnc

yaroslavVl, 2018-06-22:
hmm, yeah, strange condition.
Why is the issue closed?
`#if defined(USE_SPINDLE_DIR_AS_ENABLE_PIN) && !defined(CPU_MAP_ATMEGA328P)
  #error "USE_SPINDLE_DIR_AS_ENABLE_PIN may only be used with a 328p processor"
#endif``
Based on this condition, we will never be able to use this functionality with STM32F103C8T6 board. However, in the rest of the code this is assumed. I suggest to comment it out at all)))