usbcnc/grbl #44: error:2  annoyed

Sombat4t, 2018-05-12:
Every thing works fine and very nice.  there are only error:2 and error:11 are annoyed. 
By Universal gcode sender shows error:2. So try writing small program in C for only reading serial data to see what, find error:2 and error:11 in my reading loop. these quite weird for me.  I try to find what the cause of these problems, but cannot position them. 
Can any one help?

lakeroe, 2018-06-25:
Not sure but maybe one of these errors is caused by this: https://github.com/usbcnc/grbl/issues/39

Sombat4t, 2018-06-28:
My issue is closed now. 
I find the solution. The real problem is no synchronization mechanism in tx- ring buffer between two threads: usb interrupt and serial write in main thread.  the "serial_tx_buffer_head" is not properly handle.  this makes sending corrupted char random.

closed 2018-06-28 by Sombat4t