usbcnc/grbl #47: Nothing happens after connecting the Bluepill's USB to the computer

classbproject, 2018-06-14:
Hi, I was able to successfully build and upload the HEX file to the Bluepill. However, after I plugged in the USB cable to the computer, nothing happened. There was no new COM port that I could see. Am I missing something?

classbproject, 2018-06-15:
Yes I did. I even tried different cables. Nothing. Windows doesn’t even say that new hardware was detected. I looked through the traces on the board and there seem to be 22R resistors in series with the D+ and D- lines.

yaroslavVl, 2018-06-15:
R9, R11 (22R) is ok. Some people have problem with R10 (10k), it have to be 1,5k according to the standard.
Because of this in some cases, the board is not connected via usb. This has already been discussed here, if I'm not mistaken. It's a known bug of this board.
[image]

SailWithChips, 2018-06-15:
Hi, I had similar problems... I have experimented that If I supply voltage before and some ms after that I plug the USB then, everything worked fine for me. Can you try?