usbcnc/grbl #59: How do I compile this for STM32F103CBT6?

Alfrederson, 2018-11-07:
I got it to run just fine on STM32F103C8T6 (bluepill), but when I change the device to STM32F103CBT6 (Maple Mini) and compile it, nothing seems to happen. Blue pill shows up as a STM32 Virtual Com Port, but Maple Mini doesn't seem to do anything. It doesn't even show up as an unidentified USB device.
What do I need to do?

yaroslavVl, 2018-11-07:
try to use UART+ch340(or ftdi ft232) instead of USB by removing the definition USEUSB

siy, 2018-11-19:
Maple Mini has dedicated hardware for manipulating USB pull-up. This logic is necessary to perform re-initialization of the USB connection with host after flashing new firmware without need to perform physical disconnect/connect of USB cable. This logic might conflict with raw Bluepill firmware and leave USB in the state which does not allow host to perform proper USB initialization. Take a look into Maple booltoader code for all relevant details. My guess that it will be enough to make small update to Grbl code and leave pull-up in some predictable state. I don't remember exact value out of top of my head, but in worst case it will take only two experiments to find correct value.
Another thing which need to be changed is the size of the ROM in linker script, but this should not be a problem.

yaroslavVl, 2018-11-19:
Yes, maple mini use PB9 to control USB 1.5k pull-up resistor.
And in addition PB8 is connected to BOOT0.
So I suppose it's not a good idea to use maple mini as GRBL usbcnc controller.
Is it really necessary?