usbcnc/grbl #61: Bug in homing when an axis uses a pin between 8 and 15 for as step pin

mstrens, 2018-11-17:
When I ask for a homing, Z and Y axis are moving but not the X axis
After looking to the code, I founded a bug.
In my setup, . I use pin 10 as step pin for X axis.
But in file limit.c, in function void limits_go_home(),  there is a definition :  uint8_t step_pin[N_AXIS];
and after you have : step_pin[idx] = step_pin_mask[idx];
The issue is that step_pin_mask is an array of uint16_t instead of uint__t.
In order to let code works for several cpu, best is ti change the delaration with:
PORTPINDEF step_pin[N_AXIS];
because PORTPINDEF is already uint8_t or uint16_t depending on cpu.

mstrens, 2018-11-17:
I just tested the solution proposed above but it is not enough.
2 more changes are required:
- in the same function you must declare 
PORTPINDEF axislock ;
instead of uint8_t

- in file system.h, you must declare 
PORTPINDEF homing_axis_lock;
instead of uint8_t homing_axis_lock;

Then I tested it and it works.

yaroslavVl, 2018-11-18:
in https://github.com/usbcnc/grbl/issues/51 (https://github.com/usbcnc/grbl/files/2244605/stm32grbl11_31072018.zip) grbl.h line 79 #define PORTPINDEF uint16_t
homing work fine for all axis.

yaroslavVl, 2018-11-20:
ok! I'll add it in 4axis branch