usbcnc/grbl #62: Bug fixing to this project

mstrens, 2018-11-17:
There are several bugs reported here.
For several bugs, the fix has been provided but has not been put in the code of master branch.
Is it someone that keeps a version with all the fixes included?

yaroslavVl, 2018-11-17:
See https://github.com/usbcnc/grbl/issues/51
I've fixed some bugs and add A (B,C) axis.

mstrens, 2018-11-17:
Thanks.
Still I presume that it does not contains the last bug fixes.
Would it not be better to open a new branch/repository on github to put all your changes/fixes?

yaroslavVl, 2018-12-10:
> 
> 
> Thanks.
> Still I presume that it does not contains the last bug fixes.
> Would it not be better to open a new branch/repository on github to put all your changes/fixes?

Done! #51 
I will create a new repo, after I fix 5 and 6 axes.

Alfrederson, 2018-12-17:
@yaroslavVl I found another bug. With the code the way it is, you cannot use M3 and M5 to toggle the state of the Spindle Enable pin because it's not initialized as an output.

If you add those two lines:

 ` GPIO_InitStructure.GPIO_Pin = 1 << SPINDLE_ENABLE_BIT;
  GPIO_Init(SPINDLE_ENABLE_PORT, &GPIO_InitStructure); // <--- initializes the spindle enable pin`

To the file spindle_control.c, after the line 68, which initializes the Spindle Dir pin with the function

`GPIO_Init(SPINDLE_ENABLE_PORT, &GPIO_InitStructure);`

Then M3 Sx will turn the spindle en bit high and M5 will turn it low. That's pretty much just the missing piece. With support for 4 and 5 axis this shapes up to be a really good CNC firmware and for a really inexpensive controller.

yaroslavVl, 2018-12-17:
I'll see it.