usbcnc/grbl #63: Spindle Enable pin isn't initialized as output. Easy fix.

Alfrederson, 2018-12-17:
With the code the way it is, you cannot use M3 and M5 to toggle the state of the Spindle Enable pin because it's not initialized as an output. The fix is really simple.

If you add those two lines:

 ` GPIO_InitStructure.GPIO_Pin = 1 << SPINDLE_ENABLE_BIT;
  GPIO_Init(SPINDLE_ENABLE_PORT, &GPIO_InitStructure); // <--- initializes the spindle enable pin`

To the file spindle_control.c, after the line 68, which initializes the Spindle Dir pin with the function

`GPIO_Init(SPINDLE_ENABLE_PORT, &GPIO_InitStructure);`

Then M3 Sx will turn the spindle en bit high and M5 will turn it low, just like it was supposed to be.

yaroslavVl, 2018-12-17:
There is USE_SPINDLE_DIR_AS_ENABLE_PIN definition.
I suppose, this repeats the functionality of the native GRBL, where the number of outputs is limited. I think it's better to leave it as is. What do you think?
see line 64 of spindle_control.c for SPINDLE_ENABLE_BIT

Alfrederson, 2018-12-18:
> There is USE_SPINDLE_DIR_AS_ENABLE_PIN definition.
> I suppose, this repeats the functionality of the native GRBL, where the number of outputs is limited. I think it's better to leave it as is. What do you think?
> see line 64 of spindle_control.c for SPINDLE_ENABLE_BIT

Look elsewhere in the code. On set_spindle_status it calls a macro to change the spindle enable pin when you don't have variable spindle speed, but it does nothing because after line 64 there's a conditional define and it only configures spindle enable bit if USE_SPINDLE_AS_ENABLE_PIN is defined, and they have sepparate definitions (enable is PB 13 and dir is PB 14), so it's likely the intention was to have separate spindle enable and dir controls. Some brushless spindles have an input to change the rotation direction, and with an H-bridge a regular DC motor can be reversed as well.

When you have variable spindle enabled, the spindle dir can be toggled with M4, and the spindle pwm pin can be set to different duty cycles.

yaroslavVl, 2018-12-18:
Maybe.
You can still use spindle PWM at 100% and 0% as spindle enable bit))
Ok, we can add something like USE_BOTH_... (defined default) to use both spindle en and dir.
It is important to leave some free I/O to use it as cnc controlled I/O by M62-66 (linuxcnc style) codes at future.

Alfrederson, 2018-12-18:
> Maybe.
> You can still use spindle PWM at 100% and 0% as spindle enable bit))
Kind of. The relay will hum if I do that.

yaroslavVl, 2018-12-19:
Why? 0% is full off, 100% is full on. It is equal to digital output  off/on.

Alfrederson, 2018-12-19:
Just think of it this way: there's a pin definition for a spindle enable pin, which is different from the spindle dir. If the code is kept the way it is, the spindle enable pin will never be toggled, regardless of #defines, because it isn't ever initialized as an output. Does that sound like an intented feature?

The relay hums with an S value of anything other than 100. Yes, always use M3 S100, but how about the fact that the pin does nothing when reading through the comments would indicate it does?

Blocheren, 2023-06-25:
I have this issue, and I can't figure it out. I'm on the verge of buying a blackbox.. 🤷😅🥴