usbcnc/grbl #65: Corexy

Sa6k0, 2019-02-10:
Hello is it possible to setup grbl so it can run core XY as it is one of the reasons to move to 32 bit.
If it isnt what would be the best alternative for 32bit core XY.
Also it seems that the cocox ide is gone.
Thank you!

misan, 2019-02-10:
COREXY is supported in 8-bit grbl too.

Sa6k0, 2019-02-10:
I know but I would like to setup grlb on STM32F103 for corexy.
How would I do this?

misan, 2019-02-10:
You just uncomment this line:
https://github.com/usbcnc/grbl/blob/edge/grbl/config.h#L198

Sa6k0, 2019-02-10:
Did that but I get an avr/io.h: No such file or directory error  even when not running corexy.
I am compiling with the arduino ide.

yaroslavVl, 2019-02-10:
Did you really try to compile coide project with arduino ide?

Sa6k0, 2019-02-11:
I did because the coocox one is gone.

yaroslavVl, 2019-02-11:
I've moved this project to the atollic truestudio with some changes. https://github.com/yaroslavVl/6-AXIS-USBCNC-GRBL

Alfrederson, 2019-02-22:
I tried to compiled this in Platformio but I have no idea how to make a config for it based on the Atollic project files. I compiled it with no issue in Atollic, but sooner or later it will be gone too