usbcnc/grbl #67: USB SOFTWARE IDENTIFIER ISSUE

AxelGUILLAUMET, 2019-10-10:
Dear friends.

I bought two $3 STM32 BluePill and one ST-LINK V2 Clone on AliExpress.

After reading and studying about the STM32 technology, I started to practice...

I installed on the PC those aplications:

STM32 ST-LINK Utility (To flash .HEX and .BIN files)
STM32CubeMX (To pregenerate projects and define I/Os clock & buses)
STM32CubeIDE 1.0.2 (To write and compile C programs for the STM32)
So first I made a simple blinking led project with STM32CubeIDE, I had some debugging issues with the ST LINK V2 Clone but I could fix thanks to these informations there:

https://www.eevblog.com/forum/beginners/unexpected-idcode-flashing-bluepill-clone/

After I tried to compile those GRBL STM32 firmwares with STM32CubeIDE and NO PROBLEM, it compiles, it loads BUT... NOT USB/Virutal serial port detected :(

I tried to download the HEX files but... same there... Windows error 43, USB not identificated...

I tired with the arduino USB bootloader, same...

I looked about the 10K resistor error, changed it for a 1.8K SMD, no FIX. I resolded the micro USB conector of the board, NO FIX :(

But, I found a strange firmware there:

https://jeelabs.org/2017/09/enabling-usb-on-a-blue-pill/

I download the BIN located in the part 3 of this page, pass the jumper form 1 to 0, push reset, connect the micro USB and here we go! The bluepill is DETECTED AS "STMicroelectronics Virtual COM Port"!!!

This firmware is the only .BIN I found that have the working USB.

So it isn't an HARDWARE Problem this is a firmware ISSUE...

Here goes a pic of the STM32 of my bluepill. I saw on the web there is a lot of boards with fakes STM... I'm not able to identify an original or a fake. (The pic is in attachement to have a good resolution and see well the engraved characters)

I have these informations about it:

20:32:39 : Device ID:0x410
20:32:39 : Device flash Size : 64KBytes
20:32:39 : Device family :STM32F10xx Medium-density

Question 1 the MAIN QUESTION : I think there is something I'm making BAD or I forgot into my flashing configuration... and I don't have the experience and the sufficient knowledge to find the problem.

Regards

ewidance, 2019-10-10:
Hi Axel,

Since the author of USBCNC don't seem to be interrested anymore in that project, i suggest you to have a look on y repo where i've adapted usbcnc to be compiled with Vscode / Platformio. You'll find here : https://github.com/ewidance/grbl-usbcnc-stm32

You should also have a look on the 6 axis port : https://github.com/robomechs/6-AXIS-USBCNC-GRBL . Many bugs have been fixed.  I've also ported it form Vscode / platformio just for my own use: https://github.com/ewidance/6-AXIS-USBCNC-GRBL.

Both have been successfully complied and works on my lasercutter.

AxelGUILLAUMET, 2019-10-10:
Thanks al lot dear friend!

I will pass now on your repository! See you there!

closed 2019-10-10 by AxelGUILLAUMET

AxelGUILLAUMET, 2019-10-11:
Salut mon Ami, cela me fait plaisir de voir qu'en france il y a encore du
talent et des gens motivés pour faire de l'electronique! Je suis de Dijon
et de Barcelone et cela fait... 6 ans que je suis expat au Mexique bref
bonne aventure!

Donc je me permet de t'expliquer et c'est avec grand enthousiasme que je le
fait car tu a fait exactement ce que je veut faire : Controler une
decoupeuse laser co2 de 40W et surtout gagner en vitesse car la gravure en
arduino UNO c'est lent... trop lent... et le STM sa m'a l'air de la balle!

Je travaille les machines laser en GRBL pour avoir la plus grande facilité
avec le soft tout en 1 appelé LightBurn  https://lightburnsoftware.com/ je
pense que tu doit connaitre, bon c'est payant mais ce sont les types me
semble t'il de GRBL officiel qui font ce soft ou l'un des partenaires
principaux et franchement sa marche bien! (pour le moment je l'ai pas
acheté je charge juste la demo pour mes essais mais une fois que sa marche
je pense investir car franchement ce que coute le soft c'est de la rigolade
comparé a d'autres....)

Malheuresuement je n'ai pas pu te contacter sur github car l'option ISSUE
n'aaparait pas dans ton FORK https://github.com/ewidance/grbl-usbcnc-stm32 ....
Dommage...

Bon, pour ma part, j'ai installé l'extension PlatformIO et j'ai recompilé
avec succes le projet, c'est pour cela que je te joint el fichier .ELF afin
que si tu puisse verifier que le binary fontionne sur une de tes blue pill
et si surtout le micro-USB le PC le detecte comme un port COM... Moi pour
ma part c'est erreur 43, meme pas d'ID USB, comme si l'USB ne s'active pas
dans le STM32...

Je pense que je fait une erreur, je n'ai pas encore les connaissances
suffisantes en STM32, il doit maquer quelque chose, un bootloader ou
autre....

Le truc bizzare c'est que ce firmware -
https://git.jeelabs.org/jcw/embello/src/branch/master/explore/1608-forth/suf/usb-common.bin
-
Si fonctionne et si le PC reconnait bien l'USB de la blue pill a la
premiere! ... Du coup pas de probleme de hardware...

Je ne sais pas ce que tu en penses ou si tu a deja eu un probleme similaire
a celui-ci, sincerement je ne sais plus trop comment m'orienter pour
trouver la panne a part analyser les trames USB a l'osciloscope mais
malheureuseument mon vieil oscillo est mourant... Et ce sera assez dur...
Mais identifier si il y a de la com, sa par contre c'est possible!

Grandes salutations et bonne chance pour le FAB LAB de montpeul'!!!! C'est
beau les gens!!!!