usbcnc/grbl #70: cannot use coolant and mist

IhsanFuadi3, 2020-10-16:
Hi,why coolant and mist didn't work?
thanks

terjeio, 2020-11-08:
Because the pins are used for the JTAG interface per default? It has to be disabled first by calling `__HAL_AFIO_REMAP_SWJ_NOJTAG();` somewhere in the startup sequence. If you use JTAG for programming this is no longer possible after this call has been issued. SWJ programming should still work.

You may want check my grblHAL driver for [STM32F103](https://github.com/terjeio/grblHAL/blob/master/drivers/STM32F1xx/Src/driver.c#L940) to see how I have done it.

IhsanFuadi3, 2020-11-16:
> Because the pins are used for the JTAG interface per default? It has to be disabled first by calling `__HAL_AFIO_REMAP_SWJ_NOJTAG();` somewhere in the startup sequence. If you use JTAG for [...]

im sorry,but i dont understand what do you mean,i just learned in MCU

terjeio, 2020-11-16:
The JTAG interface is used for programming and when enabled it blocks the use of the pins for other purposes. I had to add code to unblock them before they could be used for coolant and mist.

IhsanFuadi3, 2020-11-16:
oke terjeio, thank you,i will try